bofh69/nfc2klipper #3: Android app to write NFC tags

V-aruu, 2024-08-16:
Hey there !

I really like your project, and have been using it to manage my spools.

I created a small Android app to write tags from your phone.

It's working quite good for me so far.

GitHub link : https://github.com/V-aruu/SpoolCompanion

PS : Sorry to contact you here on issues for your project, but I didn't find any other means.

bofh69, 2024-08-18:
Looks really great!

I'll add a link in the readme.

spuder, 2024-12-13:
A few of us are working on a 3rd party RFID reader called openspool. It works with bambu printers. I'm glad to see someone is doing something similar with klipper

https://github.com/spuder/OpenSpool

There is a discord where several filament manufactures such as Polar Filament are working to standardize on an open format for RFID tags called OpenTag (Formerly called OpenRFID). Many other manufactures have expressed interest in adding tags. 

I'd love to see nfc2klipper support the purposed standard. 
https://github.com/Bambu-Research-Group/RFID-Tag-Guide/blob/main/OpenSourceRfid.md

Would you be interested in joining the standard discussions on discord?

bofh69, 2024-12-18:
@spuder Please don't add unrelated comments to issue, create new ones instead. I created one for you here #8 

@V-aruu There is a link in the README now so I'm closing this issue.

closed 2024-12-18 by bofh69