bofh69/nfc2klipper #8: OpenSourceRfid

bofh69, 2024-12-18:
A few of us are working on a 3rd party RFID reader called openspool. It works with bambu printers. I'm glad to see someone is doing something similar with klipper

https://github.com/spuder/OpenSpool

There is a discord where several filament manufactures such as Polar Filament are working to standardize on an open format for RFID tags called OpenTag (Formerly called OpenRFID). Many other manufactures have expressed interest in adding tags. 

I'd love to see nfc2klipper support the purposed standard. 
https://github.com/Bambu-Research-Group/RFID-Tag-Guide/blob/main/OpenSourceRfid.md

Would you be interested in joining the standard discussions on discord?

_Originally posted by @spuder in https://github.com/bofh69/nfc2klipper/issues/3#issuecomment-2542339251_

bofh69, 2025-01-03:
nfc2klipper would have to create the vendor if missing, then the filament if missing finally the spool if missing.
The vendor & filament can be found by searching for the exact same data as in the tag.

The spool id from Spoolman will probably have to be handled by adding a new custom field in Spoolman. That field can contain the tags' serial number.

The tag writing functions would have to update the spool in Spoolman with the new serial number after writing.

Aasikki, 2025-04-07:
I think there's a bit of and egg vs chicken problem here 😆 Manufacturers might think it won't make sense to add tags before something supports reading them, and vise versa you might think it's not worth implementing before there's spools out there that have tags.

kens96, 2025-04-27:
I am not sure if my use case is unique. 
I am using Filaman scale https://github.com/ManuelW77/Filaman to read and write NFC tags, this uses OpenSpool format and stores the NFC reference in Spoolman
I was thinking of using nfc2klipper with a reader attached to the printer to read the NFC tag to tell klipper wich filament is loaded... until I released that nfc2klipper is not using the same NFC format and so this wont work

bofh69, 2025-05-08:
@spuder

Both OpenSpool and OpenTag formatted tags should be usable with nfc2klipper & Spoolman now, it not create a new bug report.

When/if OpenTag is specified to the point someone can actually implement it, I might add support for writing them from the Spoolman data.

When manufacturers start selling spools with such tags, I might add a feature to import their data into Spoolman.

Until then, I close this issue. Feel free to open it again or create new issues if either of those things happen.

closed 2025-05-08 by bofh69