bofh69/nfc2klipper #12: Multi-printer support

thijstriemstra, 2024-12-19:
Afaik only a single printer is supported right now, it's not to possible to send a spool change to a printer other than the one defined in the config.

bofh69, 2024-12-22:
Yes, the idea was to have a nfc-reader at the printers spool holder to automatically handle changes as you load the new filament.

What is your idea and workflow here?

Having a single nfc-reader where you tap the spool before mounting it on one of the printers? In that case some more input is needed to the system. Perhaps a tag that represent the printer the spool is moved to? That could also be used when storing a spool to update the location in spoolman.

Support more nfc-readers from the program? Then it is perhaps better to add an environment variable for choosing the configuration file and then just run it once per nfc-reader?

thijstriemstra, 2024-12-24:
I bought a roll of ntag215 stickers and made magnetic tags that attach to my spoolbox: [image]

at first i thought of making a rpi-based handscanner but this would need a long cable or battery pack. Since my printers are on same table, a scanner that is mounted to the wall is another idea since i can easily take out the magnetic tags, scan it and place it back.

The OpenSpool project uses an esp32 to control the np532 device and use wifi to transmit the data. These esp32 devices are much cheaper and smaller, more suitable than an rpi IMO.

So what about making a wall-mounted scanner, using esp32 (and maybe even micropython instead of c++)?

I would prefer if we also use YAML instead of json config files, which all klipper related projects seem to use.

Klipperscreen, for example, allows you to define an x nr of printers in the config, something like this:

```
[printer ender]
moonraker_host: ender.lan

[printer prusa]
moonraker_host: prusa.lan
```

bofh69, 2025-01-01:
Nice setup!

I don't think nfc2klipper would be the right place for such a feature/workflow. Its probably better to add klipper/spoolman support to the OpenSpool project instead. They've already got the hardware and a lot of the firmware needed for it.

My klipper doesn't use YAML. It is a custom format that looks like toml/Windows INI-files. Switching to toml instead of json5 should be pretty straight forward and its probably a good idea to be more similar to the wider klipper ecosystem.

bofh69, 2025-03-30:
Have a look at this project, it might be what you are looking for:
https://github.com/ManuelW77/Filaman

closed 2025-04-02 by bofh69