bofh69/nfc2klipper #13: RFID Tag only read once then get stuck in while loop.

scoggins, 2025-01-01:
Since the new web server code has been added and the new lib/nfc_handler code was introduced. I can only read 1 RFID tag before the code gets stuck in the while loop here https://github.com/bofh69/nfc2klipper/blob/f284d2a66fe546fd7dc1224dcd536951403fe948/lib/nfc_handler.py#L113 I'm not sure what's the point of this while loop. But the value of tag is never updated so it's trying to reuse the 1st tag every read. If I comment out this while loop, things work and I can read multiple RFIDs and not just the first one. 

I think the code was to try and stop the newly written tag from being read? but the clf.connect() never returns unless it can read a tag. so this is an infinite while loop that blocks on clf.connect()

@thijstriemstra  What was the original purpose of this while loop?

bofh69, 2025-01-01:
Hi!

Thanks for reporting this.

Yes, the idea was that that program should wait there for the current tag to be removed so `self._read_from_tag` shouldn't be called over and over again. However the connect method never returns with None so it gets stuck in that inner while loop now.

Just removing that while block works, but it will cause a lot of logging about reading the same spool & filament over and over again.

I'll make a proper fix for it.

bofh69, 2025-01-01:
The `bugfix/13` branch has a fix. Does it work for you too?

scoggins, 2025-01-02:
Thats working with my setup, so looks good.

bofh69, 2025-01-02:
Perfect! I've added it to main now.

closed 2025-01-02 by bofh69