bofh69/nfc2klipper #26: FilaMan interoperability

bofh69, 2025-04-28:
> I am not sure if my use case is unique. 
> I am using Filaman scale https://github.com/ManuelW77/Filaman to read and write NFC tags, this uses OpenSpool format and stores the NFC reference in Spoolman
> I was thinking of using nfc2klipper with a reader attached to the printer to read the NFC tag to tell klipper wich filament is loaded... until I released that nfc2klipper is not using the same NFC format and so this wont work 

 _Originally posted by @kens96 in [#8](https://github.com/bofh69/nfc2klipper/issues/8#issuecomment-2833594130)_

kens96, 2025-04-28:
This is brilliant, thank you! It will take me a few days to hook up the NFC board to the pi and test it out but I will feedback.

bofh69, 2025-04-29:
@kens96 
Thanks! I'll have to update the format of the id before it will work

kens96, 2025-05-04:
I Need a bit of help please my Linux skills are average at best ;) Apologies for posting this here but I wasn't sure where else to 

I cloned the repo (changed to the branch) and got the card reader working without issue but the service wont start 
it is looking for the reader on tty:AMA0 despite me having updated in the config

[image]

[image]

nano ~/.config/nfc2klipper/nfc2klipper.cfg
[image]

According to the config in /etc/nfc/devices.d/pn532_uart_on_rpi_3.conf  it is on /dev/ttyS0

[image]

Have I made a typo in the config, where can I find the nfc2kilipper log?

bofh69, 2025-05-04:
@kens96 

It first looks tries `~/nfc2klipper.cfg`, then `~/.config/nfc2klipper/nfc2klipper.cfg`. You can add:
```python
            print(f"Reading config from {cfg_filename}")
```
on line 37, the line after `with open(cfg_filename ...`  to see which file is read, but I'd guess there is one in ~/nfc2klipper.cfg that is being read.

kens96, 2025-05-04:
Great, thank you.
I removed ~/.config/nfc2klipper/nfc2klipper.cfg and edited in ~/nfc2klipper.cfg and things are working now.
I will let you know how I get on...

kens96, 2025-05-04:
Hi, @bofh69 
The log is below for a Filaman tag, it is reading the tag but attempting to do a lookup of the ID in a decimalised format, but Filaman is storing it in Spoolman as Hex 
Please let me know if you need anything further

NFC ID
4:dd:80:1c:3f:61:80

```
May 04 20:47:50 E3v2 python3[35698]: 2025-05-04 20:47:50,265 INFO - lib.nfc_handler: Read other record: NDEF Record TYPE 'application/json' ID '' PAYLOAD 110 byte '7b22636f6c6f725f6865' ... 100 more
May 04 20:47:50 E3v2 python3[35698]: 2025-05-04 20:47:50,265 DEBUG - nfc2klipper: Fetching data from spoolman from tags' id: 4:221:128:28:63:97:128
May 04 20:47:50 E3v2 python3[35698]: 2025-05-04 20:47:50,268 DEBUG - urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.9:7912
May 04 20:47:50 E3v2 python3[35698]: 2025-05-04 20:47:50,410 DEBUG - urllib3.connectionpool: http://192.168.1.9:7912 "GET /api/v1/spool HTTP/1.1" 200 2761
May 04 20:47:50 E3v2 python3[35698]: 2025-05-04 20:47:50,413 INFO - nfc2klipper: Did not find spool records in tag nor from its id (4:221:128:28:63:97:128) in spoolman
```

bofh69, 2025-05-05:
Oops. I forgot a `:x` on the byte formatting. I've updated it now.

kens96, 2025-05-05:
Working perfectly! 
Really appreciate you adding this compatibility, thank you! 
Will you merge this branch into Main?

```
May 05 09:33:05 E3v2 python3[3362]: 2025-05-05 09:33:05,157 INFO - lib.nfc_handler: Read other record: NDEF Record TYPE 'application/json' ID '' PAYLOAD 110 byte '7b22636f6c6f725f6865' ... 100 more
May 05 09:33:05 E3v2 python3[3362]: 2025-05-05 09:33:05,158 DEBUG - nfc2klipper: Fetching data from spoolman from tags' id: 4:dd:80:1c:3f:61:80
May 05 09:33:05 E3v2 python3[3362]: 2025-05-05 09:33:05,161 DEBUG - urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.9:7912
May 05 09:33:05 E3v2 python3[3362]: 2025-05-05 09:33:05,340 DEBUG - urllib3.connectionpool: http://192.168.1.9:7912 "GET /api/v1/spool HTTP/1.1" 200 2761
May 05 09:33:05 E3v2 python3[3362]: 2025-05-05 09:33:05,345 INFO - nfc2klipper: Sending spool #19, filament #19 to klipper
May 05 09:33:05 E3v2 python3[3362]: 2025-05-05 09:33:05,346 DEBUG - urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.11:80
May 05 09:33:05 E3v2 python3[3362]: 2025-05-05 09:33:05,363 DEBUG - urllib3.connectionpool: http://192.168.1.11:80 "POST /api/printer/command HTTP/1.1" 200 2
```

bofh69, 2025-05-05:
Great! I'll clean up the commits a bit and then merge it

bofh69, 2025-05-05:
Fixed by #25

closed 2025-05-05 by bofh69