bofh69/nfc2klipper #28: Using nfc2kilipper with Happy Hare and MMU

kens96, 2025-05-06:
Hi @bofh69

When using nfc2kilipper with Happy Hare and an MMU it can happen that a filament is moved from one gate in the MMU to another. In this case nfc2klipper does not send the Spoolman data to klipper because it believes it is reading the same spool
When using nfc2kilipper with Klipper and a single filament the issues does not happen as you don't load the same filament twice.

Proposal, either always send the spool and filament to Klipper or provide a configuration switch in nfc2klipper.cfg to enable this functionally for cases where it is required such as and MMU setup.

https://github.com/moggieuk/Happy-Hare
https://github.com/moggieuk/Happy-Hare/wiki/Spoolman-Support#---auto-setting-with-rfidqr-code-reader

```
May 06 18:25:29 E3v2 python3[850]: 2025-05-06 18:25:29,312 INFO - lib.nfc_handler: Read other record: NDEF Record TYPE 'application/json' ID '' PAYLOAD 110 byte '7b22636f6c6f725f6865' ... 100 more
May 06 18:25:29 E3v2 python3[850]: 2025-05-06 18:25:29,313 DEBUG - nfc2klipper: Fetching data from spoolman from tags' id: 4:2e:9c:1c:3f:61:81
May 06 18:25:29 E3v2 python3[850]: 2025-05-06 18:25:29,316 DEBUG - urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.9:7912
May 06 18:25:29 E3v2 python3[850]: 2025-05-06 18:25:29,526 DEBUG - urllib3.connectionpool: http://192.168.1.9:7912 "GET /api/v1/spool HTTP/1.1" 200 2641
May 06 18:25:29 E3v2 python3[850]: 2025-05-06 18:25:29,531 INFO - nfc2klipper: Read same spool & filament
```

bofh69, 2025-05-08:
That's a good idea. I've added it in the #29 pull request. Please try it.

kens96, 2025-05-09:
I can confirm it works as expected. Happy Hare now takes care of the swap between gates if the same filament is transferred form one gate to another on the MMU
Config and log below. 

May I suggest you leave the print statement so it is a bit easier to see in the log what config is being used? For no reason I can explain, after a hard git reset and pull my config stated using the "~/.config/nfc2klipper/nfc2klipper.cf" location for its config 

```
args = None  # pylint: disable=C0103
for path in ["~/nfc2klipper.cfg", CFG_DIR + "/nfc2klipper.cfg"]:
    cfg_filename = os.path.expanduser(path)
    if os.path.exists(cfg_filename):
        with open(cfg_filename, "r", encoding="utf-8") as fp:
            print(f"Reading config from {cfg_filename}")  # Added print statement
            args = toml.load(fp)
            break
```

Config switch 
```
# If true, the SET_ACTIVE_* macros are called every time the tag is read.
# If false, the SET_ACTIVE_* macros are not called when the same tag is read again.
always-send = true
```
Log with multiple attempts of the same tag 
```
May 09 17:15:42 E3v2 python3[4155]: 2025-05-09 17:15:42,227 DEBUG - urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.11:80
May 09 17:15:43 E3v2 python3[4155]: 2025-05-09 17:15:43,859 DEBUG - urllib3.connectionpool: http://192.168.1.11:80 "POST /api/printer/command HTTP/1.1" 200 2
May 09 17:16:14 E3v2 python3[4155]: 2025-05-09 17:16:14,920 INFO - lib.nfc_handler: Read other record: NDEF Record TYPE 'application/json' ID '' PAYLOAD 110 byte '7b22636f6c6f725f6865' ... 100 more
May 09 17:16:14 E3v2 python3[4155]: 2025-05-09 17:16:14,920 DEBUG - nfc2klipper: Fetching data from spoolman from tags' id: 4:2e:9c:1c:3f:61:81
May 09 17:16:14 E3v2 python3[4155]: 2025-05-09 17:16:14,922 DEBUG - urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.9:7912
May 09 17:16:15 E3v2 python3[4155]: 2025-05-09 17:16:15,036 DEBUG - urllib3.connectionpool: http://192.168.1.9:7912 "GET /api/v1/spool HTTP/1.1" 200 2630
May 09 17:16:15 E3v2 python3[4155]: 2025-05-09 17:16:15,038 INFO - nfc2klipper: Sending spool #20, filament #20 to klipper
May 09 17:16:15 E3v2 python3[4155]: 2025-05-09 17:16:15,040 DEBUG - urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.11:80
May 09 17:16:16 E3v2 python3[4155]: 2025-05-09 17:16:16,671 DEBUG - urllib3.connectionpool: http://192.168.1.11:80 "POST /api/printer/command HTTP/1.1" 200 2
May 09 17:17:15 E3v2 python3[4155]: 2025-05-09 17:17:15,909 INFO - lib.nfc_handler: Read other record: NDEF Record TYPE 'application/json' ID '' PAYLOAD 110 byte '7b22636f6c6f725f6865' ... 100 more
May 09 17:17:15 E3v2 python3[4155]: 2025-05-09 17:17:15,909 DEBUG - nfc2klipper: Fetching data from spoolman from tags' id: 4:2e:9c:1c:3f:61:81
May 09 17:17:15 E3v2 python3[4155]: 2025-05-09 17:17:15,911 DEBUG - urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.9:7912
May 09 17:17:16 E3v2 python3[4155]: 2025-05-09 17:17:16,063 DEBUG - urllib3.connectionpool: http://192.168.1.9:7912 "GET /api/v1/spool HTTP/1.1" 200 2630
May 09 17:17:16 E3v2 python3[4155]: 2025-05-09 17:17:16,067 INFO - nfc2klipper: Sending spool #20, filament #20 to klipper
May 09 17:17:16 E3v2 python3[4155]: 2025-05-09 17:17:16,069 DEBUG - urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.11:80
May 09 17:17:17 E3v2 python3[4155]: 2025-05-09 17:17:17,714 DEBUG - urllib3.connectionpool: http://192.168.1.11:80 "POST /api/printer/command HTTP/1.1" 200 2

```

closed 2025-05-09 by kens96