bofh69/nfc2klipper #36: TigerTag Integration

TigerTag-Project, 2025-06-13:
Hi Sebastian,

First, thank you for your work on nfc2klipper — it’s a great bridge between physical spools and Klipper automation. I’d like to propose an enhancement that could benefit a growing number of users and filament manufacturers.

🐯 Feature Proposal: Support for TigerTag Protocol

TigerTag is an open-source RFID/NFC standard for 3D printing materials, already adopted by several filament manufacturers (eSun, Sunlu, Rosa3D, R3D, etc.). It aims to provide a universal tag format compatible with all brands of printers and tools, including Klipper setups like yours.

✅ Why support TigerTag?
	•	Offline-compatible (all info encoded on tag)
	•	Readable by any app or device — schema is fully documented
	•	Already supported by our [mobile apps (iOS & Android)](https://www.tigertag.io/)
	•	Has an open API (optional): [API Docs (Swagger)](https://api.tigertag.io/api:tigertag)
	•	Offers an updated Spoolman-compatible JSON database:
https://api.tigertag.io/api:tigertag/SpoolmanDB/materials

🔧 Suggested Implementation

The TigerTag data structure is encoded in NFC tags using NDEF text records over pages 4–12 of NTAG213 chips. You can:
	1.	Add support for decoding TigerTag fields, e.g.:

2.	Map TigerTag fields to SET_ACTIVE_FILAMENT / SET_ACTIVE_SPOOL macros or allow dynamic Spoolman matching via the TigerTag Product ID (FFFFFFFF for community-made tags).
	3.	Optionally support an “auto-write” compatibility mode, where TigerTag-encoded spools are written via Spoolman and reused natively.

🔗 Additional Resources
	•	[TigerTag Protocol GitHub Documentation](https://github.com/TigerTag-Project/TigerTag-RFID-Guide)
	•	[Spoolman-compatible material.json](https://raw.githubusercontent.com/TigerTag-Project/TigerTag-RFID-Guide/main/SpoolmanDB/materials.json)

🙏 Final Thoughts

This would allow your excellent tool to support a growing ecosystem of NFC-ready filaments with universal IDs, improving usability and reducing fragmentation.

Let me know if you’d be open to this contribution — I’d be happy to help implement or test it!

Best regards,
Benoît TigerTag

bofh69, 2025-06-14:
Hi!

That looks like a great project!

I would guess there already is basic support for it via the tag's ID matching, like how FilaMan tags are supported.

Of course it would be even better to automatically store the tag's info in spoolman for new spools. I'll take a look at it when the weather is worse unless you implement it before.

BR,
Sebastian

TigerTag-Project, 2025-06-14:
[image]